p3p/pio-nxplpc-arduino-lpc176x #1: Error: Detected unknown package 'framework-arduino-lpc176x'

keshantao, 2018-10-28:
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)

PackageManager: Installing framework-arduino-lpc176x @ >=0.0.2

Error: Detected unknown package 'framework-arduino-lpc176x'

Board name: BOARD_MKS_SBASE
Build type: build
Environment used: LPC1768

p3p, 2018-10-28:
Looks like your installation of platformio is having a bad day, could you delete the platformio cache folder `.platformio` (in your user folder on windows) and try again. Removing `.pioenvs` and `.piolibdeps` from the Marlin folder could also help give it a fresh start.

keshantao, 2018-10-28:
thanks, I am giving it a try

p3p, 2018-10-28:
Can you make sure platformio is upto date, I've just tested clean installs on windows and linux and it works as expected so currently I have to assume it's something wrong at your end, I'm just not sure what could be causing it.

keshantao, 2018-10-28:
thanks, I will give it another try this afternoon, thanks for the effort

p3p, 2018-10-28:
ah good, glad you got it sorted, remember to close issues when they are fixed.

closed 2018-10-28 by p3p