p3p/pio-nxplpc-arduino-lpc176x #5: LPC1766 support?

ultimoistante, 2019-12-11:
Hi, i'm trying to program an LPC1766 mcu, flashing it with a J-Link programmer, but i always receive a **"Writing target memory failed"** error from flash process (platformio run --target upload). My J-Link programmer is correctly working as I can flask other Cortex-M3 (STM32) mcu(s) without problems.

Could please you help me? I can give all my help, in order to create new board support for this particular mcu model.

All needed files and output are attached below.
Thank you in advance

> I've tried to add a new board definition for LPC1766 (adapting the one for LPC1768, and changing maximum flash size to 208 Kbytes [256-48]):

`
{
  "build": {
    "cpu": "cortex-m3",
    "f_cpu": "100000000L",
    "mcu": "lpc1766",
    "core": "arduino"
  },
  "connectivity": [
  ],
  "debug": {
    "jlink_device": "LPC1766",
    "svd_path": "LPC176x5x_v0.2.svd",
    "tools": {
      "cmsis-dap": {
        "onboard": true,
        "server": {
          "arguments": [
            "pyocd-gdbserver.py",
            "-t",
            "lpc1766"
          ],
          "executable": "$PYTHONEXE",
          "package": "tool-pyocd"
        }
      }
    }
  },
  "frameworks": [
    "arduino"
  ],
  "name": "NXP LPC1766",
  "upload": {
    "maximum_ram_size": 32736,
    "maximum_size": 212992,
    "protocol": "mbed",
    "protocols": [
      "jlink",
      "blackmagic",
      "cmsis-dap",
      "mbed"
    ]
  },
  "url": "https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf",
  "vendor": "NXP"
}
`

> This is the content of my platformio.ini file:

`[env:nxp_lpc1766]
platform = nxplpc-arduino-lpc176x
board = nxp_lpc1766
framework = arduino
upload_protocol = jlink
`

> and this is the full output from platformio run --target upload:

Processing nxp_lpc1766 (platform: nxplpc-arduino-lpc176x; board: nxp_lpc1766; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1766.html
PLATFORM: NXP Arduino LPC176x 0.1.0 #f6eba86 > NXP LPC1766
HARDWARE: LPC1766 100MHz, 31.97KB RAM, 208KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.1.3, toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/nxp_lpc1766/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [          ]   3.5% (used 1133 bytes from 32736 bytes)
PROGRAM: [          ]   4.2% (used 9004 bytes from 212992 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed
CURRENT: upload_protocol = jlink
Uploading .pio/build/nxp_lpc1766/firmware.bin
SEGGER J-Link Commander V6.54c (Compiled Nov  7 2019 17:05:53)
DLL version V6.54c, compiled Nov  7 2019 17:05:41

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link ARM V8 compiled Nov 28 2014 13:44:46
Hardware version: V8.00
S/N: -1
License(s): RDI,FlashDL,FlashBP,JFlash,GDB
VTref=3.261V
Target connection not established yet but required for command.

Device "LPC1766" selected.
 
 
Connecting to target via SWD
Found SW-DP with ID 0x2BA01477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x412FC230. Implementer code: 0x41 (ARM)
Found Cortex-M3 r2p0, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 002BB000 SCS
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 002BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 002BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 002BB923 TPIU-Lite
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 002BB924 ETM-M3
Cortex-M3 identified.
PC = 1FFF0BA4, CycleCnt = 00038074
R0 = 7FFF8FDF, R1 = 10000124, R2 = 00000105, R3 = 40084000
R4 = 2009C000, R5 = 00000002, R6 = 40008000, R7 = 00000000
R8 = 4002C000, R9 = 2C6F4C01, R10= A4ED0126, R11= 95FC3191
R12= 1CCE656C
SP(R13)= 10007FB8, MSP= 10007FB8, PSP= B8BF109C, R14(LR) = 1FFF0CBF
XPSR = A1000000: APSR = NzCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.
 
Downloading file [.pio/build/nxp_lpc1766/firmware.bin]...
Writing target memory failed.
 
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
 
 
Script processing completed.
 
========================= [SUCCESS] Took 1.70 seconds =========================

p3p, 2019-12-12:
It seems `Writing target memory failed.` would indicate its trying to write to outside the available flash region, 

This framework is relatively generalised but was designed for Marlin and I'm not aware of any lpc1766 boards? are you intending to use it for your own firmware? the official mbed framework will be more generally compatible.

If you would like to attach the firmware.elf and bin, or make a PR with your modifications, I can have a look to see if I can help when I have time.

ultimoistante, 2019-12-12:
Thank you for the answer. The board is an hardware already done by a customer, and since we need to reduce time-to-market, i was trying to speedup development time using arduino framework and libraries.

Anyway, I'm doing some further investigations using other frameworks and programming hardware (building a BlackMagicProbe with a BluePill board). Seems there is the same problem using mbed framework (trying to upload a simple "blink"). So, the issue is still in the flashing process.

I'll give you a report asap.

ellensp, 2020-02-24:
Although this is oldish...  I see S/N: -1 
This looks to be a clone J-LINK and SEGGER software will notice this and produce all sorts of weird errors, without actually telling you why. It seems to be "copy protection". I've even had the SEGGER software blank the chip in the J-LINK clone.