p3p/pio-nxplpc-arduino-lpc176x #9: Can't compile Marlin for LPC1769  - "WinError 161"

AMDFenics, 2021-10-04:
Hello mate, 

 sadly i can't compile Marlin anymore. i hope u can help me out.

The log says:

`Executing task: C:\Users\kev_2\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\ntpath.py", line 647, in realpath
    path = _getfinalpathname(path)
FileNotFoundError: [WinError 161] Der angegebene Pfadname ist ungültig: "\\\\192.168.100.110\\volume1\\Backup\\3D Projekte\\Zip's Software und Co\\Firmware\\BTT\\Marlin-2.0.9.1\\https:\\github.com\\p3p\\pio-nxplpc-arduino-lpc176x\\archive\\0.1.3.zip"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 237, in get_package
    if not self.test_pkg_spec(pkg, spec):
  File "C:\Users\kev_2\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 255, in test_pkg_spec
    if os.path.realpath(spec.url) == os.path.realpath(pkg.path) or (
  File "C:\Program Files\Python39\lib\ntpath.py", line 651, in realpath
    path = _getfinalpathname_nonstrict(path)
  File "C:\Program Files\Python39\lib\ntpath.py", line 601, in _getfinalpathname_nonstrict
    path = _getfinalpathname(path)
FileNotFoundError: [WinError 161] Der angegebene Pfadname ist ungültig: "\\\\192.168.100.110\\volume1\\Backup\\3D Projekte\\Zip's Software und Co\\Firmware\\BTT\\Marlin-2.0.9.1\\https:\\github.com\\p3p\\pio-nxplpc-arduino-lpc176x\\archive\\0.1.3.zip"`

I think it is related to ntpath.py....

Thanks in advance and kind regards :)

p3p, 2021-10-04:
This issue appears to be in PlatformIO rather than my LPC framework, I think Windows may not like the long pathname (or the non local storage) as the error comes from the windows api layer (WinError 161 ERROR_BAD_PATHNAME).

You can probably work around it by using a local folder, and/or making sure it has a shorter path.

AMDFenics, 2021-10-04:
Hey there, i could figure it out.

 i changed ntpath.py at line 647

"path = os.path.abspath(path)##_getfinalpathname(path)"

from "_getfinalpathname" to "os.path.abspath"

The ntpath.py file is located in the python install folder.

So this Error is not related to your work ;)

Thank you anyway :)

closed 2021-10-04 by AMDFenics