p3p/pio-nxplpc-arduino-lpc176x #11: 'SConsEnvironment' object has no attribute 'IsCleanTarget'

Domochevskiy, 2022-12-04:
Hello.

Not sure if it's this framework issue, but on Marlin-bugfix-2.1.1 brunch with config `LPC1768 (board: nxp_lpc1768; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)` I've got an error:

```
Processing LPC1768 (board: nxp_lpc1768; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'SConsEnvironment' object has no attribute 'IsCleanTarget':
  File "C:\Users\domoc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\domoc\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\domoc\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "J:\3DP\LMv2\Marlin-bugfix-2.1.x\buildroot\share\PlatformIO\scripts\configuration.py", line 229:
    if pioutil.is_pio_build():
  File "J:\3DP\LMv2\Marlin-bugfix-2.1.x\buildroot\share\PlatformIO\scripts\pioutil.py", line 9:        
    return not env.IsIntegrationDump() and not env.IsCleanTarget()
```

Issue was "fixed" with commenting `and not env.IsCleanTarget()` in pioutil.py:

```
def is_pio_build():
    from SCons.Script import DefaultEnvironment
    env = DefaultEnvironment()
    return not env.IsIntegrationDump() # and not env.IsCleanTarget()
```

After commenting it'seems all is buliding ok.

p3p, 2022-12-04:
This issue may be better directed at Marlin as that is where the problem was introduced, not sure if `env.IsCleanTarget()` is supposed to exist in the PlatformIO api but if it is I don't think it would be for the mcu platforms to implement.

p3p, 2022-12-04:
Looks like its a new PlatformIO feature only available in the dev release, shouldn't have been used yet, this commit in Marlin (https://github.com/MarlinFirmware/Marlin/commit/6d1d5e4e6a13b055e78aebbae98eda1868051522) should work around it not always being available.

closed 2022-12-04 by p3p